sundevilrobotics/urc-code #10: Write GUI (Does not have to interface with subsystems yet!)

bjrice13, 2020-07-15:
A GUI needs to be written so we can control the robot wirelessly during the competition. With minimal research, http://wiki.ros.org/rqt sounds like a plausible option.

http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics is a tutorial writing data to rqt graphs; could be handy. YouTube is a good resource too.

bjrice13, 2020-07-15:
If necessary, create new ("sub") issues - that are specific to a single issue - here on GitHub, comment the new issue numbers below, and then mark this issue as resolved.

Robin-Parmentier, 2020-11-07:
Web server based applications like http://robotwebtools.org/ are probably better for this, because it means we have way more flexibility with making the GUI do special stuff and not look janky.

closed 2021-09-17 by bjrice13